BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #2: what software did you use to flash the mod bios

z4q, 2020-06-03:
im using q-flash in bios return invalid

z4q, 2020-06-03:
bro. u commited the mlb n other. please remove the commit

BenjaminX, 2020-06-03:
https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/blob/master/Tools/EFIFlash.0.80.exe.zip

http://www.freedos.org/download/

UnsignedUnsigned BIOS need EFIFlash tool run-in DOS environments.

other way, windows 10 using GA @BIOS tool resigned a new BIOS file, upload LOGO image. it's resigned file to Q-Flash.

BenjaminX, 2020-06-03:
> bro. u committed the MLB n other. please remove the commit

yes, sorry. pls, ignore MLB.

BenjaminX, 2020-06-03:
https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/blob/master/BIOS/Signed_ModBIOS_Z390MG_CFG_SP.rom.zip
Signed version, just modified CFG/SP. Q-Flash passed.

https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/blob/master/BIOS/modBIOS_Z390M_Gaming.rom.zip
unsigned version, just using EFIFlash.0.80 tool.

z4q, 2020-06-04:
thanks bro.

closed 2020-06-04 by z4q