BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #7: Trouble with iGPU and iMessage/FaceTime

adamdaduser, 2020-07-23:
Sir, thank you for your generous sharing! 
I have a similar system compared to yours, the details are as follow:

CPU: i5-8400
Motherboard: same as yours
WiFi Adapter: BCM94360CS2
Graphics Card: same as yours
(with no Thunderbolt cards)

I used your modified BIOS, as well as your BIOS configuration, I installed macOS 10.15.5 with your EFI files(only modified the PlatformInfo part) and upgraded it to 10.15.6, it works very well! But I got some problems. It would be so nice if you can give me some advice, thank you!

1. I can login my iCloud account, even open Find My Mac. AirDrop, SideCar and Handoff work just fine, but I can not login iMessage and FaceTime.

2. In Intel Power Gadget, the GFX REQ Frequency is always 0.33, GFX AVG is always 0.00, I tried edit some films using Adobe Premiere Pro, some of the hardware decoding is not usable. Despite VideoProc shows all passed.

Before open this issue, I have tried some methods but they didn’t work.
FCPX核显独显全程满速指南
http://bbs.pcbeta.com/viewthread-1836920-1-1.html

Thank you for your attention!

adamdaduser, 2020-07-23:
[image]
[image]

adamdaduser, 2020-07-23:
[image]
It's so strange that the iGPU is actually working when I am using SideCar!

BenjaminX, 2020-07-23:
cool, I will check it again.

maybe next weekend.

BenjaminX, 2020-08-18:
@adamdaduser pls upgrade to 1.6.2 checking GFX frequency issue.

and

iMessage and FaceTime issue pls reset your SN ROM MLB number.

closed 2020-12-14 by BenjaminX