BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #8: Stuck at Apple logo

Smackerboy, 2020-07-27:
Hi, im using Bios F9G.
I have setup the bios setting andmade the bootable usb thumb drive with the necessary files and mac os.
However, im stuck at Apple logo.
Need your help on this. 
Thanks.

BenjaminX, 2020-07-27:
please change plist 'boot-arg' add -v

output more logs in the screen

Smackerboy, 2020-07-27:
[image]

BenjaminX, 2020-07-27:
Update CFG unlock BIOS, first, FreeDOS booting to DOS, and flash it.

Smackerboy, 2020-07-29:
> 
> 
> flash mod BIOS with CFG unlock. First, booting with FreeDOS and flash it.

Ok, I try.  Thanks bro!
Can i flash thru Win 10 environment with the gigabtye Bios Live application to flash this mod bios?

BenjaminX, 2020-07-29:
> > flash mod BIOS with CFG unlock. First, booting with FreeDOS and flash it.
> 
> Ok, I try. Thanks bro!
> Can i flash thru Win 10 environment with the gigabtye Bios Live application to flash this mod bios?

No, you need setup FreeDOS in U Disk and flash mod version.

adamdaduser, 2020-08-09:
> > flash mod BIOS with CFG unlock. First, booting with FreeDOS and flash it.
> 
> Ok, I try. Thanks bro!
> Can i flash thru Win 10 environment with the gigabtye Bios Live application to flash this mod bios?

Actuallly, you can. install APP Center and @BIOS, Download the mod bios. Before flashing it, you need to select a picture (it will shows on booting) to fool the APP. I flashed it in this way, and it works.

BenjaminX, 2020-12-14:
be sure to upgrade BIOS F9l version, unlock CFG and SERIAL PORT disabled

closed 2020-12-14 by BenjaminX