BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #10: about sleep / wakeup

kjacer, 2020-08-11:
Hi, Ive installed this EFI and successfully done
it looks pretty good, 
but when system sleep, screen off
and it appears it can not wake up , coz the screen can not shows but the system still working ( fan still working) 

could you know whats happen of this

AndreVizosodaCruz, 2020-10-18:
Any update? i have the same issue
@BenjaminX

BenjaminX, 2020-10-18:
> Any update? i have the same issue
> @BenjaminX

no more messages for issue details, OC for me working perfectly.

AndreVizosodaCruz, 2020-10-18:
Sleep and wake is working perfect with your hackintosh?

MalikBehloul, 2020-10-19:
BIOS

AndreVizosodaCruz, 2020-10-19:
> BIOS 

What I have to change in bios?

BenjaminX, 2020-10-19:
> > BIOS
> 
> What I have to change in bios? I done everything that is in readme

try to flash F9k beta version and CFG/Serial Ports in BIOS settings

MalikBehloul, 2020-10-19:
BIOS Settings
Advanced Mode > Settings > IO Ports > Initial Display Output > PCIe 1 Slot
Advanced Mode > Settings > IO Ports > Above 4G Decoding > Enabled
Advanced Mode > Settings > IO Ports > USB Configuration > XHCI Hand-off > Enabled
Advanced Mode > Boot > CSM Support > Disabled
(If using iMac19,1 system definition) Advanced Mode > Settings > IO Ports > DVMT Pre-Allocated > 64M
(If using iMac19,1 system definition) Advanced Mode > Settings > IO Ports > DVMT Total Gfx Mem > 256M
(If using iMac19,1 system definition) Advanced Mode > Settings > IO Ports > Aperture Size > 256MB
(Only available on Beta BIOS F9k, F9j, F9l) Settings > IO Ports > Super IO Configuration > Serial Port > Disabled
(Only available on Beta BIOS F9k, F9j, F9l) Boot > CFG Lock > Disabled
* Note: Contrary to popular belief, I've never had to disable VT-x or VT-d on any of my hacks.

For those who have GC-Titan Ridge Thunderbolt 3 PCI-e card:
Advanced Mode > Settings > Thunderbolt(TM) Configuration > Thunderbolt Boot Support > Boot Once
Advanced Mode > Settings > Thunderbolt(TM) Configuration > Security Level > No Security
Advanced Mode > Settings > Thunderbolt(TM) Configuration > Discrete Thunderbolt Configuration > GPIO3 Force Pwr > Enabled

Unlock MSR (aka unlock CFG) (NVRAM will not work with locked MSR)
Update to BIOS F9l. This will allow you to do the unlock right in BIOS.
1. Download and unZip the BIOS file.
2. Copy the BIOS file to a FAT32 formatted USB flash drive.
3. Use the Q-Flash feature in BIOS to update BIOS.
4. To be safe, do not update the Backup BIOS.
5. Re-do your BIOS settings after flashing.

AndreVizosodaCruz, 2020-10-19:
My motherboard is the z390m and not the z390m gaming. Maybe this is the problem right? Because I have all you have said

MalikBehloul, 2020-10-19:
hello look this https://dortania.github.io/OpenCore-Post-Install/universal/sleep.html#preparations  good luck

BenjaminX, 2020-10-20:
> (Only available on Beta BIOS F9k, F9j, F9l) Settings > IO Ports > Super IO Configuration > Serial Port > Disabled
> (Only available on Beta BIOS F9k, F9j, F9l) Boot > CFG Lock > Disabled

Yes, Mark

EdsonLucas, 2021-12-26:
I had the same problem, i solved removing the radeon boost kext