BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #11: Not working

gregorispielmann, 2020-08-19:
Thanks for your configuration. I have a very similar desktop configuration here, except for the vga that mine is a RX 5700 XT from AsRock. The point is that I'm having the apple logo loading screen and when complete there is only a black screen. Do you know what can be the problem? Thank you again!

BenjaminX, 2020-08-19:
Sorry, really can't clarify the issue you mentioned. My advice booting with -v mode, check output log messages.

gregorispielmann, 2020-08-19:
Sorry about the little info. I was making an OpenCore from zero, but then I found your almost ready OpenCore config with basically the same hardware, so I downloaded your folder and switched my OC folder (except for the OpenCore.efi) from my usb drive for yours and changed the data from config.plist (ROM and others) that you recommend in the readme. Is that the correct way? Sorry if I'm doing this incorrectly but is my first hackintosh.

I have the same processor and motherboard, but my vga is a RX 5700 XT from asrock.

BenjaminX, 2020-08-19:
got it, sir. You need to read this in full very carefully.
Believe me, this is the only right way.

https://dortania.github.io/OpenCore-Install-Guide/
https://dortania.github.io/Getting-Started-With-ACPI/

gregorispielmann, 2020-08-19:
Thank you @BenjaminX I will try again from zero based on OpenCore Install Guide.

uiltonlopes, 2020-08-20:
@gregorispielmann 
Friend did you get to work? I have an RX 5600 XT and I have the same problem.

gregorispielmann, 2020-08-20:
@uiltonlopes hello! No I didn't get it work! The fact is that you need a boot arg agpmod=pikera for rx 5000 gpus, but I have so many troubles that I left this for another hour.

uiltonlopes, 2020-08-20:
Realmente o problema era so esse argumento mesmo, aqui rodou perfeito com o Big Sur Beta 4

gregorispielmann, 2020-08-20:
@uiltonlopes qual sua configuração? A minha é i7 9700k, Z390 M Gaming e RX 5700 XT, poderia me passar depois como tu fez?

uiltonlopes, 2020-08-20:
Posso sim man. Minha config i5 9600k, Gigabyte Z390 M Gaming e RX 5600 XT, estou terminando de refazer aqui o pendrive para o Big Sur Beta 5 pra ja ficar na ultima versão. Me manda um email que te passo certinho a pasta EFI ja pronta, e quanto ao argumento ta faltando uma letra ali agdpmod=pikera

gregorispielmann, 2020-08-20:
Mandei email. Obrigado!

closed 2020-12-14 by BenjaminX