BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #12: in memory panic stackshot succeeded ** bytes traced

bjrjustin, 2020-10-10:
problem occurs when using OC 0.6.0 and above, and keeping restart. while 0.5.9 and lower version works well.
I'm using F9g original, CFG unlocked, DVMT prelocate 64M

BenjaminX, 2020-10-10:
how to unlocked original BIOS?

bjrjustin, 2020-10-11:
boot to grub shell
setup_var_3 0x5C1 0x00
[image]

GRUB SHELL EFI:[EFI.zip](https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/files/5361147/EFI.zip)

bjrjustin, 2020-10-11:
rename EFI/BOOT/BOOTx64-grub.efi to EFI/BOOT/BOOTx64.efi

BenjaminX, 2020-10-11:
Inside your EFI/OC/Tools folder and config.plist:
run VerifyMsrE2 tool verify CFG unlock sucessfuly

BenjaminX, 2020-10-11:
Highly recommended to try this optimized modded BIOS based on 'F9g' version. Download mod BIOS and flash it for extended functionalities (CFG unlocked, SERIAL PORT disabled).

BenjaminX, 2020-10-11:
maybe you can try this beta version first
https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/issues/13

BenjaminX, 2020-12-14:
F9l BIOS pls upgrade try it again

closed 2020-12-14 by BenjaminX