BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #13: New BETA firmware F9k

hotsticker, 2020-10-10:
Hi Bejamin, thanks for sharing your guide and OC settings.  I was able to get it running perfectly.  I discovered a new BETA firmware, version F9k and wanted to share it with you.  I was not able to flash your version to set the CFG lock, however with this BETA I was able to.

https://www.tonymacx86.com/threads/gigabyte-z390-m-gaming-build-with-working-nvram.291193/
https://www.tonymacx86.com/attachments/z390mgaming-f9k-zip.480632/

BenjaminX, 2020-10-11:
very cool, I will try it.

BenjaminX, 2020-10-12:
everything is well done, F9k beta CFG\SERIAL PORT  from BIOS settings disabled.

hotsticker, 2020-10-12:
Glad to hear that.  This version made it very convenient to update and disable the settings

BenjaminX, 2020-10-12:
Release 1.6.5 updates BIOS to F9k.

closed 2020-10-12 by BenjaminX