BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #14: Netflix in Safari not working

duylhng, 2020-10-15:
I get error S7706 during Netflix playback. DRM verified working with Amazon Prime and Apple TV.

BenjaminX, 2020-10-15:
FYI here
https://www.tonymacx86.com/threads/gigabyte-z390-m-gaming-build-with-working-nvram.291193/

Should I use iMacPro1,1 or iMac19,1 system definition?
iMacPro1,1 will allow for full working DRM but Sidecar will not work.
iMac19,1 will allow for Sidecar to work but DRM may or may not work. Please see here for possible DRM workarounds. I have no experience with these workarounds and won't be of any help.
Personally, I've been on iMacPro1,1 for quite a long time and it has worked extremely well.

If you need Sidecar, change to iMac19,1 system definition. (This will break DRM support but there are workarounds. I have not tested the workarounds enough to give any support for this.)

https://www.tonymacx86.com/threads/new-drm-enabling-method-for-systems-with-amd-gpu.288015/

duylhng, 2020-10-15:
I tried iMacPro1,1 but the system freezes and restarts like 5 seconds after logging in.

BenjaminX, 2020-10-15:
check your system log first.

BenjaminX, 2020-10-15:
https://help.netflix.com/en/node/115567
https://www.lifewire.com/netflix-error-codes-4173723

duylhng, 2020-10-15:
It was working before the Safari 14 update so I think it has to do with that. I'll wait til Big Sur comes out. I also tried removing the Netflix data solution many times with no success.

BenjaminX, 2020-10-15:
> It was working before the Safari 14 update so I think it has to do with that. I'll wait til Big Sur comes out. I also tried removing the Netflix data solution many times with no success.

sounds good, maybe Safari 14 bugs.

duylhng, 2020-10-15:
Does it work for you on Safari 14?

BenjaminX, 2020-10-15:
> Does it work for you on Safari 14?

Sorry, I don’t have Netflix service available here.

duylhng, 2020-10-15:
Oh fair enough no worries then. Have you got Big Sur working by any chance?

BenjaminX, 2020-10-15:
> Oh fair enough no worries then. Have you got Big Sur working by any chance?

no, big sur environments. Waiting for apple released GM
 version will upgrade.

MalikBehloul, 2020-10-15:
hello Big Sur works very well on this mother board i have  Beta 11.0 (20a5384 c) OC works even better than on Catalina!!!  Z390M Gaming / I3-9100F 3,60 ghz -Ram 8 go- GPU Sapphire pulse radeon RX550 with FakeID . sorry for my english very poor !! it’s a small desktop computer !!! :)

duylhng, 2020-10-15:
Did you have to change any settings in config.plist? I got the installer working but on the second reboot it just restarts and keeps doing that.

MalikBehloul, 2020-10-15:
No I have not changed the config.plist neither the drivers nor the kexts but it needs the latest version of lilu kext and whatevergreen kext I am with OC 0.6.2

duylhng, 2020-10-15:
Are you using iMac19,1 or iMacPro1,1?

MalikBehloul, 2020-10-15:
iMac 19.1.   /////  https://dortania.github.io/OpenCore-Install-Guide/extras/big-sur/

duylhng, 2020-10-15:
Maybe because I was trying to install it on a spare drive that I had because technically it should work fine.

MalikBehloul, 2020-10-15:
Ok good luck !!!

duylhng, 2020-10-15:
> Ok good luck !!!

Thanks

BenjaminX, 2020-12-14:
Big Sur seems to break the support for Apple TV+ DRM.

https://github.com/acidanthera/bugtracker/issues/1034

closed 2020-12-14 by BenjaminX