BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #15: Dual monitor not working all the time on IGPU

aspitznyc, 2020-10-26:
Hi there! Awesome build. I'm using it so far with no major problems but on my dual monitor setup.
I'm only using IGPU, and from time to time I'm getting weird problems on the video, like lines on the screen (attached image)

[image]

I'm on BIOS F9l. I also have trouble with SLEEP, but I'm trying to fix one thing at a time.

Thanks!

aspitznyc, 2020-10-30:
I was able to fix the sleep isue! But still get those random lines from time to time.

MalikBehloul, 2020-10-30:
hello try with GPU pci! IGPU may be too low for 2 screens

aspitznyc, 2020-11-03:
After fixing the sleep issue, I haven't gotten those weird lines, maybe it was something there ...

BenjaminX, 2020-12-14:
Sorry, no matching the same hardware. Can't verify and fix it for now.

closed 2020-12-14 by BenjaminX