BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #17: Black screen when trying to upgrade to BigSur

hitman3r44, 2020-11-13:
Hi,
Thanks in advance for your work. Catalina 10.15.7 work perfectly, but when I tried to upgrade to BigSur, after restart and choose mac installer (Default chose by OC bootloader), load the apple logo and then became black screen.

Tried with internal GPU also, still black. But something working by seeing the HDD Lee light, I leave it for mins, it auto restart ( thought that install the new version) but in boot loader I can't find my previous OS option, only find new macos installer.

hitman3r44, 2020-11-17:
Sorry, any find any solution?

MalikBehloul, 2020-11-18:
hello I hope you updated it lilu kext as well as whatevergreen latest version good luck

BenjaminX, 2020-12-14:
No specific details on how to fix it for now. Thanks

closed 2020-12-14 by BenjaminX