BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #21: No Big Sur on OpenCore

apraw, 2020-11-18:
Hello, 
That's not my first hackintosh but my first time on OpenCore Bootloader.
I created an USB booting key with Big Sur 15.0.1 and OC Bootloader (I edited the plist with my info as shown in the guide).

But when I boot on my USB key, I can't find the "Install macOS Big Sur", I've just 5 choices : Clean VRAM, Shutdown, Verify CFG, UEFI Shell and Reset NVRAM.

With my actual Clover build on Catalina, I can see the Big Sur installer, so I can guess my USB booting key is working, but I've certainly did something bad with OpenCore.

So if you have ideas to fix this issue, I'm sure that's a dumb thing, but I can't find an answer on the web.

Thanks in advance :) (and thanks for this complete guide).

PS : I'm on Z390M Gaming and F9l BIOS, with your recommanded settings.

MalikBehloul, 2020-11-18:
hello Apraw je crois que tu peux refaire ton usb bootable 
sudo /Applications/Install\ macOS\ Big\ Sur.app/Contents/Resources/createinstallmedia --volume /Volumes/MyVolume
Nom usb=MyVolume / formaté en étendue journalisé / guid 
il doit y avoir un prob sur la realisation de L'USB car apres tu rajoute le dossier EFI de Ben en espérant que tu as la meme config sinon ........ ca part en cacahuette bon courage

apraw, 2020-11-18:
Hello Malik, merci de ta réponse. 
J'ai déjà créé 2 fois la clé (avec cette manip) mais toujours pas l'installeur sur OpenCore. J'ai remarqué que je n'étais pas le seul à qui ça arrive.
J'ai quasi la même config, mais ça ne vient pas de là, sinon ça planterait juste pendant l'install, mais là l'installeur n'apparait même pas. J'ai bien suivi les recommandations Vanilla pour OpenCore et le tuto de Ben, mais il doit avoir un truc qui m'échappe.

MalikBehloul, 2020-11-18:
essai avec la version OC 0.6.2 a la place de la 0.6.3 il y a peut etre un bleme avec la version du bootloader !!

MalikBehloul, 2020-11-18:
moi la 0.6.2 fonctionnait avec BIG Sur

MalikBehloul, 2020-11-18:
fait gaffe ca peut aussi venir du bios !!! -------> first boot

MalikBehloul, 2020-11-18:
eesai ca sur ton usb bootable 
https://github.com/Pavo-IM/OC-Gen-X/releases

enbas89, 2020-11-26:
Hi guys, I've the same problem...how can I solve it?
Thank you so much

MalikBehloul, 2020-11-26:
hello test to replace the driver (HfsPlus.efi). by VBoxHfs.efi, or test OC 0.6.3 Debeug  https://github.com/acidanthera/opencorepkg/releases 

good luck

enbas89, 2020-11-28:
Thank you bro, all done!
I've another question:
If I don't have thunderbolt, Can I remove something in Efi folder and config.plist?

Thank you

BenjaminX, 2020-11-28:
> Thank you bro, all done!
> 
> I've another question:
> 
> If I don't have thunderbolt, Can I remove something in Efi folder and config.plist?
> 
> 
> 
> Thank you

Delete path EFI/OC/ACPI 'SSDT-TBOLT3-RP05-PORT7-GC-TITAN-RIDGE.aml' and 'SSDT-DTPG.aml' two files.

Editing config.plist file ACPI sections remove items 'Thunderbolt 3' and 'Thunderbolt Hotplug'.

MalikBehloul, 2020-11-28:
super glad you managed to find the installer I don’t know if you’re on catalina or big on both will work well benefits bro

enbas89, 2020-12-08:
> Delete path EFI/OC/ACPI 'SSDT-TBOLT3-RP05-PORT7-GC-TITAN-RIDGE.aml' and 'SSDT-DTPG.aml' two files.
>
> Editing config.plist file ACPI sections remove items 'Thunderbolt 3' and 'Thunderbolt Hotplug'.

Ok done!
Last question: How have you created SSDTs? using prebuilt, using script ssdttime or manual?
Thank you so much bro!

MalikBehloul, 2020-12-08:
hello bro look this   https://dortania.github.io/Getting-Started-With-ACPI/ssdt-methods/ssdt-easy.html

MalikBehloul, 2020-12-08:
https://dortania.github.io/Getting-Started-With-ACPI/ssdt-methods/ssdt-prebuilt.html#intel-desktop-ssdts

closed 2020-12-14 by BenjaminX