BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #22: Issue with missing sensor/temp info for CPU

jaredwalters, 2020-12-05:
Anyone else have missing CPU temp? I've double checked that VirtualSMC, SMCSuperIO, and SMCProcessor kext files are being loaded.

BenjaminX, 2020-12-06:
[image]

BenjaminX, 2020-12-06:
[image]

BenjaminX, 2020-12-06:
[image]

BenjaminX, 2020-12-06:
[image]

closed 2020-12-14 by BenjaminX

uiltonlopes, 2020-12-28:
What is this program for monitoring the temperature?

BenjaminX, 2020-12-29:
> What is this program for monitoring the temperature?

iStat Menus 
https://bjango.com/mac/istatmenus/