BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #23: 支持 Catalina 吗？我的版本还没升级

g1im2, 2020-12-06:
hi，请教一下，我现在的版本在 catalina（10.15.7），还没升级，而且是 clover 引导，不知道你这个 oc 引导之前是从 catalina 升级上来的吗？我能用这个直接替换 clover 然后升级到 big sur吗？

BenjaminX, 2020-12-06:
前提硬件一样的情况，OC 是向下可以兼容的。

最好的测试办办，用 U盘加载这个 OC EFI 自己引导测试一下

g1im2, 2020-12-06:
> 前提硬件一样的情况，OC 是向下可以兼容的。
> 
> 最好的测试办办，用 U盘加载这个 OC EFI 自己引导测试一下

我的硬件跟你的差不多，cpu是8700，显卡是蓝宝石590，主板一样，只是没尝试过efi，最近用着写代码正顺着，想着之前安装时候费了两月的时间调试驱动就有点后怕。没接触过oc，所以想请教下，也可以省却一些麻烦。不过经你这么一提示，倒是真在理。我先替换掉efi尝试一下是否能顺利开机，看还有问题没再升级也ok。感谢你的回复，我这就去把原来的efi替换测试一下。

g1im2, 2020-12-06:
有一点最值得感谢的就是看了你的项目后，发现这主板的bios升级版本出来了。把cfg开关和iwatch解锁都解决了，上星期升级了bios，确实把这两个问题都解决了，再次表示感谢。

mochadwi, 2020-12-13:
I would love to build similar same gear with yours @BenjaminX but I'm curious (translating google from the above discussion) you said that OC is backward compatible, did you mean I can use your OC EFI to runs in Catalina?

(Currently, I'm doing android development, thus the android world still not optimized for Big Sur especially the IDEs, and the same with @g1im2 to afraid to debug this too long)

MalikBehloul, 2020-12-13:
hello if you have the same configuration as Ben Board mere GPU CPU ...yes the EFI OC works on catalina and Big Sur!! by bro good luck

g1im2, 2020-12-16:
> I would love to build similar same gear with yours @BenjaminX but I'm curious (translating google from the above discussion) you said that OC is backward compatible, did you mean I can use your OC [...]

> I would love to build similar same gear with yours @BenjaminX but I'm curious (translating google from the above discussion) you said that OC is backward compatible, did you mean I can use your OC [...]

yes, you can use this efi if your motherboard is z390M gaming. but use this efi, you must change graphics model if you cpu not 9700k and imac must change with this. i debug this efi and boot up with catalina success. after that, i upgrade to big sur and no problem happened. it work better than clover with my computer.