BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #29: there is no option to select the hd in the boot menu

uiltonlopes, 2020-12-26:
friend can help me, in the boot menu does not appear the option to enter the on my hd, only others to open a terminal.

MalikBehloul, 2020-12-26:
bonjour test pour remplacer le pilote (HfsPlus.efi). par VBoxHfs.efi, ou testez OC 0.6.3 Debug https://github.com/acidanthera/opencorepkg/releases

bonne chance

uiltonlopes, 2020-12-27:
[image]
shows only those options, help me please

BenjaminX, 2020-12-27:
> [image]
> shows only those options, help me please

https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/issues/19

Link here

uiltonlopes, 2020-12-28:
this solved my problem, thank you, master.

closed 2020-12-28 by uiltonlopes