BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #56: no Ethernet and no Wifi with Version 1.7.9

hankman13, 2021-03-10:
I upgraded from version 1.7.4 to 1.7.9 and figured out, that I can't use Ethernet or Wifi.
With a downgrade to version 1.7.8 everything seems fine.

BenjaminX, 2021-03-12:
more details pls

hankman13, 2021-03-14:
Well, maybe it helps to know my hardware setup

Components | SKU 
-- | -- 
CPU | Intel i5-9700K 
Motherboard | Gigabyte Z390M Gaming mATX 
WiFi Adapter | fenvi T919
Graphics Card | RX 580 8G 
Thunderbolt Card | don't have one
SSD | nvm.e SSD 1TB 
RAM | 32 GB
Time Machine Hard Disk | 2TB SATA
2nd SSD | Dual Boot Windows

The issue itself shows as crossed out Wifi Icon. In the system settings for network ethernet and Wifi are marked not connected and they are changing positions in the sidebar.

MalikBehloul, 2021-03-14:
https://www.tonymacx86.com/threads/hackintosh-big-sur-wifi-problem-with-fenvi-t919-bcm94360cd.308591/

hankman13, 2021-05-13:
Hi,
today I updated to Version 1.8.3 and I had again the flickering with the Wifi devices. So I compared the config.plist and tried to change the order in Kernel and ticked the option DisableIoMapper.
[image]

This solved my problem so far.

closed 2021-05-13 by hankman13

vintikzzz, 2021-05-15:
Same here! After installing fresh 1.8.3 there was no Ethernet connection. Enabling DisableIoMapper fixes this.

hankman13, 2021-07-15:
@vintikzzz I am still at 1.8.3 and wanted to upgrade to 1.8.5 and I failed (No Bluetooth and no Wifi). Did you manage this step?