BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #60: Does iGPU works?

hitman3r44, 2021-04-01:
I want to run the system with my i7 9700k integrated graphics card. does it possible??

MalikBehloul, 2021-04-02:
hello you can but there are manipulations to do the graphic chipset of your processor and the Intel UHD 630 you need Lilu Kext and Whatevergreen of the folder EFI/OC/kext of opencore and configure the config.plist here is a video  https://www.youtube.com/watch? v=paEke_oobZI&t=3s&ab_channel=TechNolli !!! good luck

MalikBehloul, 2021-04-02:
hello you can but there are manipulations to do the graphic chipset of your processor and the Intel UHD 630 you need Lilu Kesxt and Whatevergreen of the folder EFI/OC/kext of opencore and configure the config.plist here is a video https://www.youtube.com/watch? v=paEke_oobZI&t=3s&ab_channel=TechNolli

hitman3r44, 2021-04-02:
Thanks let me try and let you know soon

creekgroup, 2021-04-22:
> Thanks, let me try and report you soon    > Thanks, let me try and report you soon  [Загрузка EFI.zip...]()