BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #61: USB mapping not working

z4q, 2021-04-29:
mouse and keyboard not detect.

dlopez-sic, 2021-04-29:
Same here, using 1.8.0 version.

For reference, here's the System Info for the USB Bus. (The CSR8510A10 is an USB Bluetooth).

[image]

z4q, 2021-04-29:
https://github.com/evaslr/Hackintosh-Gigabyte-Z390M-GAMING
i use this EFI temporary for now.

z4q, 2021-04-30:
ok tested. usb 3.1 red one not working. other usb work fine thanks @BenjaminX

BenjaminX, 2021-04-30:
USB gen 3.1 work fine.
check again.

[image]

dlopez-sic, 2021-04-30:
Just installed 1.8.2 and now is fixed.

Thank you @BenjaminX

z4q, 2021-05-04:
thanks @BenjaminX work fine now

closed 2021-05-04 by z4q