BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #62: If SMBIOS is changed, does this affect the USB Port mapping?

hotsticker, 2021-05-08:
Hi, thanks for the latest release.  Everything is working for me except for USB 3.0 ports.  I set the SMBIOS to MacPro7,1 because it fixes DRM playback issues.  Does a different SMBIOS change the USBPorts.kext no longer function?

hotsticker, 2021-05-10:
Ok, thanks for the tip @BenjaminX!

atl285, 2021-05-10:
> SMBIOS and USBPorts.kext, no direct relationship between them.

SMBIOS has a direct relationship to `USBPorts.kext`. As You can see in the embedded `Info.plist` file, the SMBIOS is part of the USBPorts.kext:
[image]

So if You change the SMBIOS, You have to change the IOKitPersonalities also to the new SMBIOS. In this case you have to change `iMac19,1-XHC` to `MacPro7,1-XHC`.

hotsticker, 2021-05-10:
Ah, that's very helpful to know!  Thank you for the insight and pointer to where to make the update, @atl285!!

BenjaminX, 2021-05-11:
> SMBIOS has a direct relationship to `USBPorts.kext`. As You can see in the embedded `Info.plist` file, the SMBIOS is part of the USBPorts.kext:
> [image]
>
> So if You change the SMBIOS, You have to [...]

cheers