BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #63: Titan Ridge Freeze in MacOS 11.3 and 11.3.1?

LiveSorcerer, 2021-05-14:
Hello, you could get me one doubts?
You commented that you had to remove the Titan Ridge TB3 Card, as it is giving Kernel Panic.
Is the symptom the moment you enter the system?
When I upgraded to this version and on the last reboot, the installation stopped for the remaining 7 minutes, and I had to manually reset it.
And after that the system keeps locking up all the time.
In the error that appears, it appears to be USB, but the USB Map was made based on this motherboard.
But I have the impression that it is the Thunderbolt 3.

hotsticker, 2021-05-14:
I had this same problem also.  It would always lock up and I thought the system needed to be started off fresh again.   However, I got it to work properly again after I ran Disk Utility in Recovery Mode on the drive and "Repair" on the drive.  I'm not sure why that would have any effect, but afterwards, the installation continued.  It got stuck again and then I would reboot manually and then run the Disk Utility repair again.  It finally completed the update and is now functioning properly!  I don't know why it is, but the system is ok now

LiveSorcerer, 2021-05-14:
What is your setup?
Here I installed on different SSDs to see if the same problem occurred, but the issue still continues.
Follow my setup:
Intel - Core i9-9900K 3.6 GHz 8-Core Processor 
Noctua - NH-U12A CPU Cooler 
Gigabyte - Z390 M GAMING Micro ATX LGA1151 Motherboard 
Corsair - Vengeance LPX 64GB (4 x 16 GB) DDR4-3200 Memory 
2x Samsung - 970 Evo 500 GB M.2-2280 Solid State Drive 
5x Seagate Ironwolf 4TB HDD 
Gigabyte Radeon RX VEGA 56 8 GB Video Card 
GC-Titan Ridge Thunderbolt 3 
2x Dell UltraSharp U2419H 
1x LG 32” 4K UK550 
Corsair - RM850x  80+ Gold Certified Fully Modular ATX Power

BenjaminX, 2021-05-15:
Yes, TB3 card caused this problem after 11.3, and not enough time to solve this problem temporarily. It seems that the firmware and ACPI patches of TB3 need to be modified.

LiveSorcerer, 2021-05-16:
I bought a 120GB SATA SSD to use in the System version test, and it really is Titan Ridge that is freezing the 11.3 and 11.3.1 system.
The way is to stay on 11.2.3 until they fix it on MacOS 11.4 or firmware that supports the new versions.
Another problem that I'm having in OC 0.6.9, it recognizes my Intel Bluetooth, also the Dongle, but both don't work, I'll stay in 0.6.8 for now.

LiveSorcerer, 2021-05-17:
I don't know if you saw this link:
https://www.tonymacx86.com/threads/success-gigabyte-designare-z390-thunderbolt-3-i7-9700k-amd-rx-580.267551/page-1640#post-2087524

BenjaminX, 2021-05-19:
> I don't know if you saw this link:
> https://www.tonymacx86.com/threads/success-gigabyte-designare-z390-thunderbolt-3-i7-9700k-amd-rx-580.267551/page-1640#post-2087524

cool I will try it

PCIe card V1.0 (before Jul 2020)
Big Sur 11.3 onwards requires TitanRidgeMacOSFirmware.bin

zzto1118, 2021-05-20:
Hi, I think this may be a problem with the 11.3 system firmware. It seems that Mac Pro has a similar problem
[https://developer.apple.com/forums/thread/676392]

I also had to shut down forcibly every half an hour on 11.3.1. But my attempt to upgrade to 11.4 beta seems to have completely solved this problem. It has been working continuously for 24 hours at 11.4

Thanks

LiveSorcerer, 2021-05-31:
I flashed my Titan Ridge for this firmware and works.
Updated to 11.4 and Thunderbolt 3 works with hot plug.

BenjaminX, 2021-06-08:
> I saw you updated to MacOS 11.4 and OpenCore 0.7.0, did you ever change Titan Ridge firmware?

Recently, no time for testing, and I don't plan to get TB3 card for the time being. maybe next month.

LiveSorcerer, 2021-06-08:
Notice any changes between Opencore 0.6.9 and 0.7.0?
Here I'm already testing MacOS Monterey, I'm finding it fast, but it has bugs.

astroboyvn88, 2021-07-19:
This seems to be a weird question but is there a way to disable the Titan ridge card in macOS (maybe via DSDT, can you show how to achive this)? The only thing I need TB3 is to connect my eGPU RTX card in windows for 3D rendering. 

Unmodified Titan ridge card seems to make my macOS freezes after 2, 3 mins I boot up my hackintosh

atl285, 2021-09-14:
Downgrade to BIOS version F9l will bring back Thunderbolt functionality with Titan Ridge card in combination with Samsung EVO 970 NVMe SSD under Big Sur:
[image]
No freezes, no reboots! It just work!

LS-BO, 2021-09-14:
> 降级到 BIOS 版本 F9l 将带回带有 Titan Ridge 卡的 Thunderbolt 功能以及 Big Sur 下的三星 EVO 970 NVMe SSD： 不卡顿，不重启！它只是工作！
> [image]

老铁，官网没有找到F9I版本呀，这个版本在哪里下载呢

atl285, 2021-09-14:
@LS-BO, You can go back in this repo to the release 1.7.1 from 6th January 2021. This release was the last, which contains the [mb_bios_z390-m-gaming_f9l.zip](https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING/blob/c0847f44e81ba9a08d10e2bdb4f78a43bb567446/BIOS/mb_bios_z390-m-gaming_f9l.zip).

LS-BO, 2021-09-14:
> @LS-BO, You can go back in this repo to the release 1.7.1 from 6th January 2021. This release was the last, which contains the [...]

非常感谢，我用你的方法，系统再也没有出现问题，系统用起来很顺畅！！！

acaibowl1, 2023-01-06:
@atl285 I have the same mobo (same bios version) and TB card. I can't get passed the "Press <DEL> to enter setup." screen.

When I use the Displayport port everything works fine and MacOS/OpenCore boot up. Am I missing something?

atl285, 2023-02-06:
> @atl285 I have the same mobo (same bios version) and TB card. I can't get passed the "Press ~ to enter setup." screen.~

Sorry I don't understand, what the problem is?