BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #66: Bluetooth not working

hitman3r44, 2021-07-09:
Hi,

version 1.8.4 my Bluetooth isn't working 
version 1.8.5 my can't boot, after loading apple logo can't see any graphics  
version 1.8.3 everything is working

Do you need any kind of dump file, let me know how to make a dump and send it here

hankman13, 2021-07-15:
Hi @hitman3r44 ,
I have also issues with my bluetooth on 1.8.5.
For the version 1.8.3 I needed to tick the option "DisableIoMapper", to get Wifi running, but this not working in 1.8.5.
#56 
Maybe you can provide your hardware configuration.

hitman3r44, 2021-07-16:
Hi @hankman13 

Thanks for the reply,
Here is my Hardware information:

CPU: i7-9700K
GPU: ASUS Radeon HD 7790 Bonaire XT
RAM: Corsair Vengeance LPX 32 GB (2 x 16 GB) DDR4-3200 CL16 
WIFI Card: MQUPIN T919 Wireless PCI Wifi Card BCM94360CD 
(https://www.amazon.ca/MQUPIN-Wireless-Continuity-BCM94360CD-Beamforming/dp/B08JQ6GPGM/ref=pd_sbs_1/143-9491205-3233520?pd_rd_w=pxagC&pf_rd_p=4dc33e2e-16b5-4e12-aab9-e86d5748e0cb&pf_rd_r=9PHS9G41965SP7EY0GM5&pd_rd_r=df830347-635f-4172-a238-f78be608212e&pd_rd_wg=caVG4&pd_rd_i=B08JQ6GPGM&psc=1)
SSD: Samsung 970 Evo 500 GB M.2-2280 NVME Solid State Drive

Can you give me some tutorial where I can able to help in this project. Something like how to use hackintool, OpenCore Configuration or any other tools, thanks

hankman13, 2021-07-16:
Well, right now I am working with try and error.
On your graphics issues, I don't know how good your graphics card is supported.

But the Fenvi T919 we have in common with the same issue. I think it has something to do with the usb-mapping.
So maybe I will try the USBPort.kext from Version 1.8.3. Or I have to get a clue how to use hackintool to build my own usb mapping (https://www.youtube.com/watch?v=uww6Cizil_g)

hankman13, 2021-07-23:
I tried to update to 1.8.6 and found out, that is have to use the usbkext.kext from version 1.8.3.

hitman3r44, 2021-07-23:
So we needed to use old kext, in new version?

hitman3r44, 2021-07-25:
@hankman13 Hi, I tried with  1.8.6 with 1.8.3's usbkext.kext, its working fine for me. Thanks

hankman13, 2021-07-28:
Well, as I understood the usbkext.kext is responsible for the usb mapping. Somehome the new version doesn't match with my hardware.

hitman3r44, 2021-07-28:
Got it is there any tutorial for usbmapping for hackintool?

hankman13, 2021-07-29:
Well on youtube you find a lot e.g. https://youtu.be/uww6Cizil_g

bissoqui, 2021-09-22:
After the installation the bluetooth didn't work either, so I changed the file "USBPorts.kext" from version 1.8.3 and after restarting it worked perfectly.

kansonkong, 2022-01-15:
Sir, has the bluetooth issue been fixed with OC 0.7.7?

kansonkong, 2022-01-15:
您好，请问OC 0.7.7版本解决蓝牙不稳定的问题了吗？谢谢