BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #70: Can't access Opencore boot

mrsachou, 2021-11-15:
Hello, i have follow your guide and use your EFI but it does'nt boot on Opencore Menu , its stuck at the Power menu where i can go into bios, i have try to go direct to boot menu but same its stuck

Pain-and-Love, 2021-12-31:
same error.
i stuck at the GIGA LOGO  after choose  UEFI

Pain-and-Love, 2021-12-31:
> Hello, i have follow your guide and use your EFI but it does'nt boot on Opencore Menu , its stuck at the Power menu where i can go into bios, i have try to go direct to boot menu but same its stuck; I have Z390 M Gaming - Intel Core i9 - 56 vega Thanks for your help

oh~, i resolve it,  u should change the  ROM/MLB/etc.  in the config.plist  before  use the EFI

hankman13, 2022-09-29:
I have similar error. I have a dual boot setup with 2 ssds connect. One is the Hackintosh, the other in Win11.
Sometimes I can boot easy to the OS Picker and everything is fine. But often it happens, that I am stuck at the Giga Logo and nothing is happening.