BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #81: bluetooth does not work

ShawLocke, 2022-08-31:
Components | SKU
-- | -- 
CPU | Intel i9-9900
Motherboard | the same
WiFi Adapter | BCM94360CD (4 antennas) 
Graphics Card| the same

Everything works except bluetooth. Bluetooth is always off, it does not respond when I turn it on. System report shows bluetooth is not driven.

My question is: need to do usb mapping? Thanks.

ShawLocke, 2022-08-31:
I found a interesting thing. In Windows, the USB hubs are HS10 and HS12. 
[image]

However, the customized USB ports in macOS is HS11.
[image]

@BenjaminX is this the reason why 94360cd bluetooth did not get driven?

ShawLocke, 2022-09-04:
If anyone found BT is not working, please do below: 
1. find `/EFI/OC/Kexts/USBPorts.kext`, right-click it and choose `show package contents`, and drag `Info.plist` to desktop.
2. use propertree to open above file, type `command + F` to search `HS11`, change `HS11` (2 places) to `HS10` or `HS12`, port value will be `0A000000` (corresponds to `HS10`) or `0C000000` (corresponds to `HS12`). Notice: take a look at which motherboard usb port you are connecting.
3. save the file and drag it back to original place.
4. reboot.

[image]

@BenjaminX 
I think you should mention bluetooth may not work

closed 2022-09-05 by ShawLocke

hongxiangfei, 2022-09-09:
Thanks OP. I met the same problem and have solved it with this method. One more thing, remember to clean NVRAM during reboot : )

BenjaminX, 2022-09-09:
I will check it again, incredible.
[image]

atl285, 2022-10-24:
Hi, guys.

The connectors on the board contains 2 USB ports (each of the pin rows are one port):

`F_USB1` ... contains USB ports HS09 & HS10
`F_USB2` ... contains USB ports HS11 & HS12
 
This aren't USB hubs.