BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING #84: Kernel panic during installation

fiveohfiveoh, 2023-01-23:
Hi, thanks for this repo. I have the same CPU and Mobo and a Sapphire Radeon RX580 8GB GPU. 

I have tried the latest OC 0.8.8 EFI from this repo and keep getting a kernel panic during install - I tried Monterey and Venutra and get the same kernel panic on each. The kernel panic is: `Panic(CPU 5, time 2913048261358): NMIPI for unresponsive processor: TLB flush timeout, TLB state:0x0`

Installation starts (with gray background), then the system reboots to a white Apple logo on  black background. The installer says "29 minutes remaining" and the kernel panic always happens at "25 minutes remaining". 

I have OC 0.5.8 working with Catalina very well for 2 years. I have updated my BIOS to F9 and the settings match the BIOS settings stated in this repo. I have also tried OC 0.8.5 with an EFI and config.plist I generated manually using the Dortania guide. 

Has anyone else encountered this problem or know what the solution could be? Thanks!

maitpatni, 2023-06-10:
Hi,

I am also having the same issue with Rx580 and this mobo.
Earlier it was working on bios f9l.
With the latest Bios f9 I am getting this issue.
Any solution?
Thanks.

MalikBehloul, 2023-06-10:
MAJ  Opencore 0.9.2 ;)

maitpatni, 2023-06-10:
I am on opencore 0.9.2 only

MalikBehloul, 2023-06-10:
F9 BIOS from GIGABYTE resolves the Apple Watch unlock(SERIAL PORT disabled) issue and provides the CFG Unlock in the BIOS. CFG Unlock is required for this EFI to work properly. Be sure to upgrade F9.

MalikBehloul, 2023-06-10:
https://github.com/BenjaminX/Hackintosh-Gigabyte-Z390M-GAMING

maitpatni, 2023-06-10:
BIOS is on F9 only and I am facing this kernal panic issue after bios update. EFI is also latest one from this Repo.
Earlier on F9l it was working.

MalikBehloul, 2023-06-10:
CFG Unlock is needed for this EFI to work you have to enter in the bios and put CFG on Unlock and update the kextsLilou	 1.6.5	
AppleALC 1.8.2	
VirtualSTM 1.3.1	
Whatever the green 1.6.4	
NVMeFix 1.1.0	
IntelMausi 1.0.7